ut-robotics/picr22-team-saga #7: [Mechanics] Thrower assembly too wide

Akustav, 2022-09-24:
The thrower assembly is 0.4mm too wide and tabs clip into the bottom plate:

[image]

maidlam, 2022-09-30:
Current distance between thrower walls is 50,40 mm.
The top and bottom plate have different distances for the holes for the walls.

[image]
[image]

The holes in the top plate are sketched separately, while the bottom plate uses a pattern to make the holes, that's where the offset might be coming from.

[image]

Don't know if the holes on the actual robot might have been modified, or the thrower was just jammed in.

Akustav, 2022-10-01:
I think we had one robot that had alignment issues due to a oversight in the review. I think it was milled out like that. I think we modified the thrower tabs to accommodate the thrower. Seems like a small issue, overall.

closed 2022-10-01 by Akustav