ut-robotics/picr22-team-saga #15: Incorrect encoder connector pinout

ReikoR, 2022-10-01:
[image]

All existing encoders use this pinout:

1. GND
2. 3.3V
3. ENCA
4. ENCB

Power needs to be connected. Right now it has not been connected.

AnishShr, 2022-10-03:
Conenctor with correct wire orientation updated.

ReikoR, 2022-10-04:
Avoid placing GND symbols pointing up. 
[image]

GND should point down and positive power symbols should point up. It would be nicer to use power symbols for them not the net labels.
[image]

Encoder connectors are still connected to GND2. Encoders need to be connected to MCU GND.

AnishShr, 2022-10-10:
changed the symbol orientations accordingly.

closed 2022-10-11 by ReikoR