ut-robotics/picr22-team-saga #18: PDRV8243HRXYQ1 does not have SPI

ReikoR, 2022-10-01:
[image]

DRV8243 drivers we have are the HW variant in VQFN-HR (14) package.

AnishShr, 2022-10-03:
Symbol and footprint with VQFN-HR(14) variant updated

ReikoR, 2022-10-04:
All configuration pins cannot be left unconnected.
[image]

**IPROPI** pin needs a resistor to set the output current limit. I recommend configuring current limit to around 2 - 3A to help prevent motor from overheating.
[image]

**DRVOFF** need to be connected to ground, otherwise the outputs will be disabled. DRVOFF has internal pullup.
[image]

**DIAG** pin I would recommend connecting to GND to allow the driver to enable the outputs again after a small off period (t<sub>off</sub>). This allows for higher motor power, but it might be ok without it also.
[image]
[image]

**MODE** pin needs to be connected to set the driver to PH/EN (PWM/direction) mode.
[image]

I added older datasheet that contains pre-production driver information to the components page:
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/electronics/components.html#_motor_driver

AnishShr, 2022-10-06:
According to the datasheet:   
For the VQFN-HR package, AIPROPI = 3070 A/A   
We are operating the HW variant in RLVL6of6 for ITRIP, SR, and DIAG, meaning not connecting those pins.   
For this config, VITRIP = 2.97 V   
For this configuration:   
ITRIP Regulation = VITRIP / RITRIP * AIPROPI   
This yields RITRIP approximately to 3040 Ohms. What will be the closest resistance we have in the 0603 package to this resistance?

ReikoR, 2022-10-08:
You might have already checked yourself, but in the lab we have 3k and also 2.7k and 3.3k 0603 resistors.

AnishShr, 2022-10-10:
Connected all other pins as instructed in the datasheet.

closed 2022-10-11 by ReikoR