ut-robotics/picr22-team-saga #19: DRV8243 VM pins need more capacitance

ReikoR, 2022-10-01:
[image]

More capacitance is needed during switching at driver's input (VM) to avoid voltage drop at the driver's output.
Around 50 uF or more might be fine. For example it's possible to use 5x 10 uF ceramic capacitors.

AnishShr, 2022-10-03:
50uF capacitance put across the driver's output

ReikoR, 2022-10-04:
[image]

50uF 0603 capacitors do not exist.

For example 47uF 25V capacitors are much bigger and expensive:
https://www.mouser.ee/c/passive-components/capacitors/mlccs/multilayer-ceramic-capacitors-mlcc-smd-smt/?capacitance=47%20uF&voltage%20rating%20dc=25%20VDC&instock=y&sort=pricing

Using 5x 10 uF 1206 25V+ capacitors or similar is better option.

Typical recommendation is to use capacitors with voltage rating at least twice the maximum working voltage. For 16.8V batteries it the closest voltage rating is 35V, but 25V should also be ok in this case. Ceramic capacitor capacitance decreases with voltage.

[image]
https://www.murata.com/en-us/support/faqs/capacitor/ceramiccapacitor/char/0005

AnishShr, 2022-10-10:
changed the 50uF capacitor with a parallel combination of 5 10uF capacitors (5 X10uF = 50uF)

ReikoR, 2022-10-11:
10uF capacitance is bit too big for 0603 capacitors. 0603 10uF capacitors are not that common and cost more. In the lab we have 1206 capacitors. We can also order 0805 capacitors which cost about the same as 1206 ones. 

You should change 10uF capacitors to either 1206 or 0805 package.

AnishShr, 2022-10-11:
changed the 10uF capacitors to 0805 package

closed 2022-10-12 by ReikoR