ut-robotics/picr22-team-saga #20: Keep isolated sides completely separate

ReikoR, 2022-10-01:
There needs to be clear separation between isolated sides, that splits the board in two.

Illustrative image with some modifications. Taken from [this guide](https://www.ti.com/lit/an/slla284f/slla284f.pdf?ts=1664620687565):

[image]

ReikoR, 2022-10-03:
Yes, there should not be any copper.

AnishShr, 2022-10-10:
Isolator's sides to be isolated are completely isolated (Not even copper pour in between)

ReikoR, 2022-10-11:
Grounds should not extend past the isolators.

[image]

AnishShr, 2022-10-11:
GND and GND2 filled on both inner layers with ground not extending past the isolators

closed 2022-10-12 by ReikoR