ut-robotics/picr22-team-saga #22: Mounting holes are too close to the edge

ReikoR, 2022-10-01:
[image]

AnishShr, 2022-10-03:
Mounting holes shifted a little inside.

ReikoR, 2022-10-05:
Holes are still probably too close to the edges:
[image]

Also Typical M3 bolt head has a diameter of about 5.5mm. There should not be any tracks or components near the mounting holes that might be damaged by the bolt head:
[image]

AnishShr, 2022-10-10:
Mounting hole's external diameter is set to 6mm, and the hole to 3mm.

closed 2022-10-11 by ReikoR