ut-robotics/picr22-team-saga #23: Add reverse voltage protection MOSFET to motor drivers power input

ReikoR, 2022-10-01:
More info:
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/electronics/components.html#_reverse_voltage_protection

ReikoR, 2022-10-04:
Yes, they are connected.

AnishShr, 2022-10-10:
Added the reverse voltage protection MOSFET to Motor driver's input pin.

ReikoR, 2022-10-11:
1. There is no need for multiple MOSFETs, 1 is enough.
[image]

2. I don't think that 1N53xxB diodes exist in 0603 package. You need to select a diode that we have in the lab (not sure if and what we have) or that can be ordered. For BUK6Y10 you probably won't need the diode, because the gate-source voltage rating is 20V, which is higher than the battery voltage.
[image]

3. You definitely should not have 10-ohm resistor across battery voltage.
[image]

In the [guide](https://components101.com/articles/design-guide-pmos-mosfet-for-reverse-voltage-polarity-protection) they are used a a load. For the mainboard the load is the motors.
[image]

AnishShr, 2022-10-11:
changed the connections as instructed with just one MOSFET

closed 2022-10-12 by ReikoR