ut-robotics/picr22-team-saga #24: Use as wide as possible connections for high currents

ReikoR, 2022-10-01:
Motor driver outputs and power supply should use wider connections.

Based on Saturn PCB Toolkit a 0.4 mm wide track can carry 2.3A at 20 degree temperature rise:

[image]

Motor drivers might output more current and power supply traces need to carry 3 times of that if drivers are connected in series.
I recommend having wide and short copper zones for high currents.

AnishShr, 2022-10-03:
Motor driver's output with track width 0.5mm updated.

ReikoR, 2022-10-05:
[image]

It would be better if connections were shorter and wider.
Current pin headers are not a very good choice for connecting motor outputs. It would be better to create component with a SMD pad footprint large enough to allow wires to be soldered to it.

For example something like this:
[image]

Small vias can be placed inside drivers pads for the shortest connections to power. Capacitors can be placed on the other side of the board below drivers.
[image]

AnishShr, 2022-10-10:
Track width for the motor driver's output lines have been set to the width of the motor driver's ouptut pad width.

AnishShr, 2022-10-11:
changed the connectors making the connections short

ReikoR, 2022-10-12:
[image]

High current connections need to be wider.

Use copper zones and multiple small vias (0.2/0.45mm is the recommended smallest size by JLCPCB).

Rough example:
[image]

Driver outputs and V+ should also be connected with wide copper fills.

AnishShr, 2022-10-13:
High current paths all connected with copper fills

ReikoR, 2022-10-15:
1. Use solid pad connection for high currents:
[image]
[image]
[image]

2. Use multiple small vias instead of one larger via. Multiple small vias have smaller total inductance, higher current carrying capability and don't pull away the solder that easily when placed inside a pad.
[image]

3. Drill diameter for small vias should be 0.2mm, otherwise the annual ring will be too narrow.
[image]

4. Use larger pads for high currents. Current pads might be too small to solder wires to. Larger pads also are more robust. Smaller pads might break off the board. Putting multiple vias inside the pad would also help to make it mechanically more robust.
[image]

AnishShr, 2022-10-20:
All power lines routed through copper fills

ReikoR, 2022-10-23:
1. Reverse voltage protection MOSFET could be rotated to make VM polygon wider.
[image]
[image]
\
It would be even better if battery input (V+) pad could be placed on top of the MOSFET.
[image]

2. Motor driver VM capacitor could be better aligned with the driver pads.
[image]

3. There is no need to place vias inside pads for motor driver configuration pins. This will make soldering harder.
[image]
\
Something like this should be enough:
[image]

AnishShr, 2022-10-23:
[image]

How can we get rid of this issue? The copper fill clearance is set to 0.25 mm, but the fill has a clearance of 0.1016mm only in one of the edges.

ReikoR, 2022-10-23:
You should use different priority levels for adjacent zones. Then the zones are filled in an order and clearance between them is automatically kept.
[image]

AnishShr, 2022-10-23:
[image]
Is this warning fired because 'VM' is present in only one layer?

ReikoR, 2022-10-23:
Yes

AnishShr, 2022-10-23:
So shall I copper fill on the other layer as well? Or what shall I do?

ReikoR, 2022-10-23:
Just remove those vias, since they are not connecting to anything on other layers.

AnishShr, 2022-10-23:
These vias also include the ones in the decoupling capacitors. Should I remove them as well?

ReikoR, 2022-10-23:
If you mean the ones connected to MCU VDDA, then yes.

AnishShr, 2022-10-23:
unnecessary vias removed and power routed via copper fills and usage of multiple small vias where necessary

AnishShr, 2022-10-26:
Most part of the nSleep trace routed in front copper making wise copper fills for VM

ReikoR, 2022-10-26:
[image]

Rotating the reverse voltage protection MOSFET would help to improve current paths:
[image]

AnishShr, 2022-10-26:
MOSFET rotated and shifted to provide better current distribution path as instructed.

ReikoR, 2022-10-27:
MOSFET should not moved too much to the center of the board. Having signal traces close to each other will cause more noise on them.

[image]

AnishShr, 2022-10-27:
MOSFET moved away from other traces

ReikoR, 2022-10-29:
You should also route tracks straighter and with more spacing between them. Noise on a trace from an adjacent trace is bigger when tracer are closer to each other or are close for longer. Running traces parallel next to each other should be minimised.
[image]
[image]

AnishShr, 2022-10-30:
Spacing between traces increased

closed 2022-10-31 by ReikoR