ut-robotics/picr22-team-saga #26: Use switching regulator instead of linear regulator for servo power

ReikoR, 2022-10-01:
Linear regulators are too inefficient for higher currents.

ReikoR, 2022-10-05:
[image]
1. On the schematic you have TPS62932, not TPS62933.

2. Pinout is incorrect. Correct pinout from datasheet:
[image]

3. Power inductor needs to connected to SW pin, not a ferrite bead. We have SRP6060FA-5R6M inductors, which are suitable.
[image]

4. Switching regulator and all other components connected to the motors power should be on the other side of the isolators. PCB needs to be clearly divided into two isolated sides.
[image]

AnishShr, 2022-10-10:
The footprint for ferritebead changed to indcutor. And the isolators have been placed between two different power sources, dividing the PCB into two parts in regard to the power source.

ReikoR, 2022-10-11:
1. Bootstrap capacitor needs to be connected between BST and SW pins. SW pin needs to be connected to inductor.
[image]

2. Feedback resistors need to connect to FB pin.
[image]

AnishShr, 2022-10-11:
changed the connections for the BUK6Y10. Removed the diode and 10 ohm resistors, with feedback resistors connected to FB pin, and bootstrap capacitor between SW and BST pin

ReikoR, 2022-10-15:
1. Inductance is still incorrect.
[image]

2. Try to follow layout guidelines in the [TPS62933 datasheet](https://www.ti.com/lit/ds/symlink/tps62933.pdf).

For example:
> Place the inductor, input and output capacitors, and the IC on the same layer.

Currently regulator and inductor are on the top layer and capacitors on the bottom layer.
[image]

> Place the input and output capacitors as close as possible to the IC.

Output capacitor is far away.
[image]

> The VIN and GND traces must be as wide as possible and provide sufficient vias on them to minimize trace impedance. The wide areas are also of advantage from the view point of heat dissipation.
Keep the SW trace as physically short and wide as practical to minimize radiated emissions.

Current traces/connections are narrow and could probably be shorter. Use multiple small vias (0.2/0.45) for VIN, GND, +5VD for lower inductance. In general multiple smaller vias work better than one larger via for reducing inductance and having higher current carrying capability.

3. It would also be better to have servo connector near the regulator output.
[image]

4.  In regards of more general layout, having isolators on the top layer and switching regulator components on the bottom side might improve the layout. Right now every signals needs a via on both sides of the isolators.
[image]

AnishShr, 2022-10-20:
Inductor connection updated, with input/output capacitors placed close.

Also, the power lines are routed through copper fills, each pad with multiple small vias.

ReikoR, 2022-10-23:
1. Inductance value is still incorrect.
[image]

2. From [datasheet](https://www.ti.com/lit/ds/symlink/tps62933.pdf):
> Place the feedback divider as close as possible to the FB pin. A > 10-mil width trace is recommended for heat dissipation. Connect a separate VOUT trace to the upper feedback resistor. Place the voltage feedback loop away from the high-voltage switching trace. The voltage feedback loop preferably has ground shield.

It's better to have separate 5V feedback trace that is shielded with ground vias. Running it on opposite layer (top in this case) should help to shield it more.
[image]

3. All the vias marked with a yellow arrow do not connect to any other layers and should be removed:
[image]

4. There is only 1 ground via:
[image]
\
Something like this would be better:
[image]
I rotated and moved the capacitors a bit and added copper fill on bottom layer and multiple ground vias.

AnishShr, 2022-10-23:
8.8 uH Inductor placed with power routed through copper fills and use of vias

ReikoR, 2022-10-25:
1. SRP6060FA-5R6M is a 5.6uH inductor, not 6.8uH.
[image]

2. 5V feedback trace is quite close to driver signal traces.
[image]
\
It would be better to increase distance between traces and add ground vias for shielding:
[image]

3. 5V feedback should only be connected with trace on top layer and not with copper fill on bottom layer:
[image]

4.  FB trace is very close to the inductor.
[image]
\
Try to follow layout recommendations in the datasheet:
[image]
\
Something like this might be better:
[image]

AnishShr, 2022-10-26:
Changed the layout and added GND vias across one side of the 5V trace.

closed 2022-10-26 by ReikoR