ut-robotics/picr22-team-saga #31: Fix DRC errors (and warnings if possible)

ReikoR, 2022-10-12:
Use **Inspect -> Design Rule Checker**

Unconnected nets are the probably the biggest problem:
[image]

Unconnected nets can also be seen without running DRC.
[image]

AnishShr, 2022-10-13:
DRC errors fixed, except few(Not an error actually)

Please have a look

ReikoR, 2022-10-15:
There is a clear unconned +3.3V net:
[image]

AnishShr, 2022-10-15:
The unconnected 3.3V nets both are connected to two 3.3V pins of the microcontroller, which are internally connected. Is it necessary to connect them externally?

ReikoR, 2022-10-16:
Power should never be routed through components. Component might not have enough current carrying capability and all the noises add together if current paths are shared between components. Ideally power should be routed from single point (power source) to each component through separate (straight) path.

It would probably be better to change bottom layer copper fill from GND to +3.3V and route most of the signals on top layer to reduce gaps in the +3.3V copper. There already are 2 GND copper fills on internal layers.

ReikoR, 2022-10-23:
There are still a lot of errors that can be fixed.

AnishShr, 2022-10-23:
DRC errors fixed (Few warnings left: especially for mounting holes vias)

closed 2022-10-26 by ReikoR