ut-robotics/picr22-team-saga #32: Improve USB signal routing

ReikoR, 2022-10-15:
Current routing:
[image]

1. USB data signals are a 90-ohm differential pair and should be routed on the PCB to match that.
2. Using vias should be avoided for differential pairs or if not possible, should be kept symmetric in both signals in a pair, so that same amount of vias is used and at the same locations in both signal traces.
3. Traces should also be as symmetric and straight as possible. 
4. Differential signal routing should usually prioritised over routing other (single-ended) signals and power.

JLCPCB has [impedance calculator](https://cart.jlcpcb.com/impedanceCalculation) for their stackups.
[image]

We'll use JLC3313 stackup, because it has smaller gap between layers, which helps to improve signal integrity because electric fields (between signal and reference [usually GND] layer) of different signals interact less with each other, since electric fields don't spread out as much.

8 mil gap and 6.03 mil trace widths are close enough to 2mm and 1.5mm, so those can be used.

Differential pair (DP) width and gap can be specified in the Board Setup.
[image]

Some board constraints should also be adjusted.
[image]

Default shortcut for differential pair routing is 6.
[image]

AnishShr, 2022-10-16:
Should the differential pair be symmetric for connections to the ESD protection as well, or is it ok if the differential pair routing is symmetric only from the microcontroller pins to the SUB?

ReikoR, 2022-10-16:
Ideally differential pair should be symmetric everywhere. ESD protection should be aligned so that the signals can be routed underneath the ESD protection component.

Example layout of ESD protection component with same pinout as CM1223-02SO:
[image]

Image taken from here:
https://www.we-online.com/catalog/media/o109030v410%20AppNotes_ANP002_TheProtectionOfUSB20Applications_EN.pdf

AnishShr, 2022-10-20:
Differential pair routed

ReikoR, 2022-10-23:
[image]

USB differential signals can be more straight and symmetric.

KiCad differential pair routing does not work very well, so some manual routing and moving of tracks is probably needed.

Improved result can look something like this:
[image]

AnishShr, 2022-10-23:
Differential pair for USB routed appropriately (please check)

ReikoR, 2022-10-25:
1. USB signal traces do not have correct width:
[image]

2. I'm not sure if there are manufacturing issues with it, but it's probably good idea to avoid sharp (internal) corners:
[image]
\
Connect straight to pads if possible:
[image]
\
Or move traces slightly away from the pads:
[image]

AnishShr, 2022-10-26:
Traces palces straight to the USB pads in uC.

ReikoR, 2022-10-26:
USB signal trace width is still incorrect.

AnishShr, 2022-10-26:
USB signal trace changed to 0.153 mm

ReikoR, 2022-10-27:
[image]

Spacing between traces needs to be 0.2mm (not 0.25mm) to have correct differential impedance.

AnishShr, 2022-10-27:
spacing between the USB differential traces maintained to 0.2 mm

closed 2022-10-29 by ReikoR