ut-robotics/picr22-team-saga #33: Place decoupling capacitors as close as possible to power pins

ReikoR, 2022-10-15:
[image]

[image]

ReikoR, 2022-10-23:
MCU decoupling capacitors are incorrectly placed or can be closer to MCU pins:
[image]

AnishShr, 2022-10-23:
Decoupling capacitors placed closer and in correct place

ReikoR, 2022-10-25:
1. Each power pin should have one 100nF capacitor close to it:
[image]

2. Decoupling capacitors should be connected on the same layer with wide traces. Connecting them through bottom layer with vias will have worse inductance which decreases the usefulness of the capacitors.
[image]
[image]

AnishShr, 2022-10-26:
All decoupling capacitors connected with wide traces in same layer

ReikoR, 2022-10-26:
1 uF capacitor on the schematic is meant for analog supply filtering:
[image]
[image]

AnishShr, 2022-10-26:
1 uF capacitor placed in appropriate position.

closed 2022-10-27 by ReikoR