ut-robotics/picr22-team-saga #34: Unnecessary vias

ReikoR, 2022-10-23:
1. There are a lot of vias connected only on 1 layer. This can also be checked with DRC.
[image]

2. You don't need that many vias for low currents. Up to 2 per decoupling capacitor is enough. Adding to many vias inside pads can make soldering hard.
[image]
\
Something like this would be good for most decoupling capacitors:
[image]
Keeping +3.3V and GND vias close to each other makes sure that electric field between them spread out less.

3. Programmer connector does not have any high currents. 1 via per pin is enough.
[image]

AnishShr, 2022-10-23:
Unnecessary vias removed

AnishShr, 2022-10-23:
Can we put just one via in the Isolators pins and decoupling capacitors?

ReikoR, 2022-10-26:
1. Another thing to avoid are wide gaps in copper fills created by vias that are close to each other:
[image]
\
More space should be added between vias:
[image]

2. Use same small size via everywhere on the board:
[image]

AnishShr, 2022-10-26:
Uniform via size maintained all over the PCB with gaps maintained as well.

closed 2022-10-27 by ReikoR