ut-robotics/picr22-team-saga #39: Routing improvements

ReikoR, 2022-10-27:
1. Rotating servo connector would improve routing.
[image]
\
[image]

2. Thrower and servo signals could be swapped to get rid of 2 vias in servo signal.
[image]

ReikoR, 2022-10-29:
On the MCU side, try to route mostly on top layer, so that there is more room for (3.3V) current to flow. Right now current (indicated with yellow lines) has to go through narrow gaps or around traces.
[image]

AnishShr, 2022-10-30:
[image]

Is this much routing in the back copper considerable?

AnishShr, 2022-10-30:
New design with minimal routing (as minimum as possible) in back copper layer in the uC side updated.

ReikoR, 2022-10-31:
New routing is better.

closed 2022-10-31 by ReikoR