ut-robotics/picr22-team-saga #60: Mainboard documentation

ReikoR, 2023-01-14:
1. _"There is a SUB connector in the board"_
\
SUB -> USB

2. Specify what type and how many motors the board can control and what is the input voltage and output current limits.

3. Programmer pinout is incorrect.
\
[image]

4. Thrower connector has 4 pins.
\
[image]

5. Add reference designators to connector pinouts.

6. Typos/misspellings:
    - regualtor
    - motros
    - sneds
    - Conenctors

AnishShr, 2023-01-14:
Errors corrected.

ReikoR, 2023-01-15:
1. _"The board can controlcontrol 3 DC motors."_
\
controlcontrol -> control
\
Specify which type of DC motors. Brushed or brushless.

2. The output current of the Voltage regulator is about 2.77 A.
\
Do you mean motor driver, instead of voltage regulator?
Also 2.77A is not correct if 3k IPROPI resistor is used.

3. All encoder connector pinouts are written as Motor1.
\
[image]
\
Encoder connector pinouts can be grouped together, since the pinouts are the same.

4. README.asciidoc needs to be in `electronics` directory as is written here:
https://ut-robotics.github.io/picr22-home/team_documentation_requirements.html

AnishShr, 2023-01-15:
3.3k resistor is sued in IPROPI pin, so the output is around 2.77 A. This is explained in the updated file.

AnishShr, 2023-01-16:
All changes made as requested and pushed to repo.

closed 2023-01-18 by ReikoR