ut-robotics/picr22-team-saga #62: Team documentation

ReikoR, 2023-01-15:
1. [README.asciidoc](https://github.com/ut-robotics/picr22-team-saga/blob/documentation/README.asciidoc) comments here https://github.com/ut-robotics/picr22-team-saga/pull/61

2. Instructions on how to set up and run the code should be in [software/README.asciidoc](https://github.com/ut-robotics/picr22-team-saga/blob/software/software/README.asciidoc), not in root directory [README.asciidoc](https://github.com/ut-robotics/picr22-team-saga/blob/software/README.asciidoc).

3. Do not add robot specific information to software README, especially not passwords.

4. [firmware/README.asciidoc](https://github.com/ut-robotics/picr22-team-saga/blob/firmware/firmware/README.asciidoc) does not contain "Format and description of the data that can be sent to and received from mainboard."

AnishShr, 2023-01-15:
1. Instructions on how to setup the code moved to software/README.asciidoc
2. Robot specific information removed from software/README.asciidoc

ReikoR, 2023-01-15:
1. Added some comments to https://github.com/ut-robotics/picr22-team-saga/pull/61

2. There is no point documenting old firmware.
New firmware is waiting for a few issues to be resolved, otherwise it's functional.
https://github.com/ut-robotics/picr22-team-saga/pull/59

AnishShr, 2023-01-15:
For the new firmware, Artyom was working on it. I don't think he is working on it anymore.

closed 2023-01-18 by ReikoR